gassajor000/pn532pi #5: Failed to instance class on rpi zero

h4de5, 2020-04-14:
I do got the following error message on **raspberrypi zero**:

> TypeError: 'NoneType' object is not callable

This is probably because of a failing import of the HSU interface and the `None` initialization afterwards in 
https://github.com/gassajor000/pn532pi/blob/3b652729fec10d5cbfb817e8b519f90d76bbc30d/pn532pi/__init__.py#L4

As a workaround you can import the interface directly e.g.:

    from pn532pi.interfaces.pn532i2c import Pn532I2c

gassajor000, 2020-04-15:
Thanks for reporting this. I don't have a Pi Zero so I wasn't able to test on it. I'll look into why this is happening and figure out a fix.

gassajor000, 2020-04-15:
If you can post the error triggering the except block when importing Pn532Hsu, that would be helpful.

h4de5, 2020-04-15:
I tried to directly import them without the try catch. It seems the "serial" module was missing too. But after "sudo pip3 install serial" I get the following error message:

```
Traceback (most recent call last):
  File "./readnfc.py", line 27, in <module>
    from pn532pi.interfaces.pn532hsu import Pn532Hsu
  File "/usr/local/lib/python3.4/dist-packages/pn532pi/interfaces/pn532hsu.py", line 1, in <module>
    from serial import Serial
ImportError: cannot import name 'Serial'
```

gassajor000, 2020-04-16:
I believe the `serial` library on pypi is for serializing json. The serial COM library is called `pyserial` (which is then imported as serial). Try doing `pip install pyserial` and see if that works.

h4de5, 2020-04-17:
you are right. its working with `pyserial`
I really though that had something to do with HSU

gassajor000, 2020-04-17:
Ok cool. This should be fixed with the fix for install_requires. I'll try to get an new version uploaded today.

closed 2020-04-17 by gassajor000

h4de5, 2020-04-17:
btw. great work! :D 
before I tried hours to get the reader working using the libs directly - the pip package is a relief!

PR #7 (2020-04-17): Fix install_requires. Add typing dependency. Rev patch number. Fixes #6.
Also fixes #5