gassajor000/pn532pi #9: Remote I/O error

mandud, 2020-11-18:
Hi,

May you write up more detail about the documentation ?
I've tried run some examples, but all failed

I'm using i2c 

`pi:~/pn532pi/examples$ python3 ntag21x_rw.py
NTAG21x R/W
-------Looking for PN532--------
Traceback (most recent call last):
  File "ntag21x_rw.py", line 77, in <module>
    setup()
  File "ntag21x_rw.py", line 39, in setup
    versiondata = nfc.getFirmwareVersion()
  File "/home/pi/.local/lib/python3.7/site-packages/pn532pi/nfc/pn532.py", line 154, in getFirmwareVersion
    if (self._interface.writeCommand(bytearray([PN532_COMMAND_GETFIRMWAREVERSION]))):
  File "/home/pi/.local/lib/python3.7/site-packages/pn532pi/interfaces/pn532i2c.py", line 56, in writeCommand
    return self._readAckFrame()
  File "/home/pi/.local/lib/python3.7/site-packages/pn532pi/interfaces/pn532i2c.py", line 157, in _readAckFrame
    responses = self._wire.transaction(reading(PN532_I2C_ADDRESS, len(PN532_ACK) + 1))
  File "/home/pi/.local/lib/python3.7/site-packages/quick2wire/i2c.py", line 78, in transaction
    ioctl(self.fd, I2C_RDWR, ioctl_arg)
OSError: [Errno 121] Remote I/O error`

I've tried using another code, works well

`pi:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- 24 -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi:~$ python3 /home/pi/nfc/raspberrypi/python/example_get_uid.py 
Found PN532 with firmware version: 1.6
Waiting for RFID/NFC card...
...Found card with UID: ['0xa3', '0x64', '0x68', '0x3']
.Found card with UID: ['0xa3', '0x64', '0x68', '0x3']
.Found card with UID: ['0xa3', '0x64', '0x68', '0x3']
.Found card with UID: ['0xa3', '0x64', '0x68', '0x3']`

gassajor000, 2020-11-18:
Remote IO error is a general i2c error thrown by the quick2wire library. I
most often saw this error when the pn532 failed to send an ack because it
had reset. Be sure your power supply for the pn532 can supply enough
current (the 3.3v pin on the raspberry pi is not sufficient).

-Jordan

OguzCiftci, 2020-11-27:
If you don't have hardware issue, try adding delay after defining the i2c mode. 

Example::
i2c = Pn532I2c(1)
time.sleep(1)
nfc = Pn532(i2c)

combateer3, 2021-01-04:
I'm also getting this same issue. `nfc-poll` works properly and I get expected results from it, but I'm getting the remote I/O error from this library too.  I'm also using the 5V pin on the pi instead of the 3.3V pin, so I'm not sure that power is the issue.
I tried one of your examples and actually it seemed to recognize the presence of the chip (in other words it passed the `getFirmwareVersion` call), but rather gives the error when I call `SAMConfig()`.  Here is the dump:
```
Found chip PN5 0x32 Firmware ver. 1.6                                                                                                                                                                                                                                                                                        Traceback (most recent call last):                                                                                                                                                                                                                                                                                             File "rfid.py", line 53, in <module>                                                                                                                                                                                                                                                                                           setup()                                                                                                                                                                                                                                                                                                                    File "rfid.py", line 20, in setup                                                                                                                                                                                                                                                                                              nfc.SAMConfig()                                                                                                                                                                                                                                                                                                            File "/home/pi/.local/lib/python3.7/site-packages/pn532pi/nfc/pn532.py", line 305, in SAMConfig                                                                                                                                                                                                                                if (self._interface.writeCommand(header)):                                                                                                                                                                                                                                                                                 File "/home/pi/.local/lib/python3.7/site-packages/pn532pi/interfaces/pn532i2c.py", line 56, in writeCommand                                                                                                                                                                                                                    return self._readAckFrame()                                                                                                                                                                                                                                                                                                File "/home/pi/.local/lib/python3.7/site-packages/pn532pi/interfaces/pn532i2c.py", line 157, in _readAckFrame                                                                                                                                                                                                                  responses = self._wire.transaction(reading(PN532_I2C_ADDRESS, len(PN532_ACK) + 1))                                                                                                                                                                                                                                         File "/home/pi/.local/lib/python3.7/site-packages/quick2wire/i2c.py", line 78, in transaction                                                                                                                                                                                                                                  ioctl(self.fd, I2C_RDWR, ioctl_arg)                                                                                                                                                                                                                                                                                      OSError: [Errno 121] Remote I/O error
```

gassajor000, 2021-01-07:
@combateer3 that is odd. Have you tried running the PN532 on it's own power supply? Depending on your setup it could still be hitting a current limit. 

If it's not a power issue then I think you would have to debug the i2c lines to see if the PN532 is not sending an ack or something. I use a cheap logic analyzer from amazon for this kind of thing. If it is completing the transaction properly then that would mean there is a bug in the quick2wire library.

siltemc, 2021-04-30:
@gassajor000 
Hello,
I am also have the same issues with this library and examples, similar to @combateer3. I am using a RP4 with a 15W power supply and the NFC is powered from the 5V rail. So it should not have any power issues.
When using the libnfc library from terminal `nfc-poll` I am able to read any NFC device I want. But I would like to do something similar with python. I have a Logic 8 from Saleae but am unable to access it until the end of next week. It is currently locked up in my office because it is Golden Week in Japan right now.
Were you able to reproduce this problem on your own? Is there any fix possible?

gassajor000, 2021-04-30:
@siltemc unfortunately without more detail about what is going on I think it will be rather hard to reproduce issue. I can try running through all the examples again this weekend but I'm fairly certain they all ran when I posted the last release. If I'm not able to reproduce the issue before you get your Seleeae, go ahead and capture the i2c lines (getting an analog read on the vcc pin might be good as well) and add it to the issue.

gassajor000, 2021-05-02:
I ran the iso14443, mifare mem dump, and readMifare examples again and they all worked (can confirm that running off the 5v pin seems to be fine). Only time I got a Remote IO error was on the read Mifare one which I think because it does loops doing the read and probably tries to go faster than the pn532 chip can respond. Which example in particular were you having issues with? 

Side note, I forgot that you need to change the interface in the examples since the default is set to UART. I'll probably change that since the README shows an I2C setup.

siltemc, 2021-05-07:
@gassajor000 I was having this same issue with every example from this project. The flags for I2C/SPI/UART were easy to find and I had already changed them before reporting I was also getting the error. But it is good to put in the readme!

This picture is the i2c using terminal and the libnfc library. It was polling every 300ms I am only show where the card was read (it is one of the cards that came with the pn532 kit I have):
[image]

This is the terminal output. The scan data for the NFC UID is D4 C3 2F 2A, you can see this in the i2c data at the end on the logic output and in the table near the bottom of the data column. 
[image]

Below is the python example for iso1443a_uid. But the output is the same for every example with python. Channel 3 is the VCC pin.
[image]
the python error is the same Errno 121 Remote I/O Error

gassajor000, 2021-05-09:
Hmm from what I can tell, the Remote IO Error was thrown because the chip didn't respond to the I2C read with an ACK. The example was just querying the firmware version though so I'm not sure why it didn't respond. Perhaps the read came too quickly after the write. You can try putting a `time.sleep(1)` on [Line 157 of pn532.py](https://github.com/gassajor000/pn532pi/blob/master/pn532pi/nfc/pn532.py#L157) and see if that makes a difference. If you still have the debugger I had also intended to monitor channel 2 (the vcc pin) with an Analog signal so I could see if it dipped low.

noukee, 2021-06-17:
I have the same issue. I have tried putting a time.sleep(1) but still not working

msp1974, 2022-07-15:
I know this is an old issue now, but I came here via the google as I was having the same issue.  I have found the fix.  It seems that the issue lies in the pn532i2c.py file in the interfaces.  In the begin() method, when you set the bus, you need to add a delay here to let the i2c bus establish before sending any commands.

So @gassajor000 advice is correct but his actually needs to be put in line 24 of pn532i2c.py.  Works every time since doing that.

Hope that helps!

chengguan, 2023-07-29:
I am doing it on an RPI4 and am facing the same issue. I punch in the command line by line and it has been intermittently giving me the firmware version.

[image]

I have also tried connecting the vcc to 5V and it is still intermittent.

chengguan, 2023-07-29:
Here is what happen when I run the example directly...

[image]

gassajor000, 2023-09-03:
Closing this issue since it is old and has been piled onto by many people with different setups. For any future Remote IO Errors please open a dedicated ticket with the corresponding template.

closed 2023-09-03 by gassajor000