gassajor000/pn532pi #10: Remote I/O error in I2C

simeoncapy, 2021-06-02:
Hello,
I have the same problem as the other issue. I tried to run the script ```FeliCa_card_read.py``` from the example folder in I2C. I'm powering the PN532 with an Arduino to have enough current (I tried in 3.3 and 5 V). But I got ```Remote I/O error```, here is the complete error message:

```
NTAG21x R/W
-------Looking for pn532--------
Traceback (most recent call last):
  File "FeliCa_card_read.py", line 120, in <module>
    setup()
  File "FeliCa_card_read.py", line 46, in setup
    versiondata = nfc.getFirmwareVersion()
  File "/usr/local/lib/python3.7/dist-packages/pn532pi/nfc/pn532.py", line 154, in getFirmwareVersion
    if (self._interface.writeCommand(bytearray([PN532_COMMAND_GETFIRMWAREVERSION]))):
  File "/usr/local/lib/python3.7/dist-packages/pn532pi/interfaces/pn532i2c.py", line 56, in writeCommand
    return self._readAckFrame()
  File "/usr/local/lib/python3.7/dist-packages/pn532pi/interfaces/pn532i2c.py", line 157, in _readAckFrame
    responses = self._wire.transaction(reading(PN532_I2C_ADDRESS, len(PN532_ACK) + 1))
  File "/usr/local/lib/python3.7/dist-packages/quick2wire/i2c.py", line 78, in transaction
    ioctl(self.fd, I2C_RDWR, ioctl_arg)
OSError: [Errno 121] Remote I/O error
```

I don't have an oscilloscope to check the signal. The PN532 is detected with the ```i2cdetect -y 1``` command.

PS: sorry to open a new issue and not completing the previous one, but I tried to post on 3 different browsers, disabling AdBlock, but I got each time the error "You can't comment at this time."

gassajor000, 2021-06-03:
Ok. As I mentioned on the other issue, this is a very general error that indicates the pn532 chip didn't respond to a command. Without a logic analyzer it's pretty hard to debug. You can do what I suggested for the other issue which is to add `time.sleep()` delays at various points to see if it's a timing issue. Other than that there's not much I can do.

simeoncapy, 2021-06-03:
I put the ```pn532``` lib in debugging to print message. It looks it fails in ```_readAckFrame``` on the first call of the ```transaction```. I tried to understand the ```quickwire.i2c.py``` class, and the function, but it's below my knowledge. But I don't know if there is a way to display more useful information from ```ioctl(self.fd, I2C_RDWR, ioctl_arg)```.

```
writeCommand: bytearray(b'\x02')    bytearray(b'')    [0, 0, 255, 2, 254, 212, 2, 42, 0]
wait for ack at :
1622713717.136005
```

Because it fails on the first call, I don't know if any delay would help. I'll see if I can find an oscilloscope.

simeoncapy, 2021-06-04:
I could make it works once! I installed ```libnfc``` from git, and I can detect with that lib my cards, for example :
```
pi@rpi:~/Documents/nfc/pn532pi/examples $ nfc-list
nfc-list uses libnfc 1.8.0
error   libnfc.driver.pn532_spi Unable to wait for SPI data. (RX)
error   libnfc.driver.pn532_spi Unable to wait for SPI data. (RX)
NFC device: PN532 board via I2C opened
NFC device: PN532 board via I2C opened
pi@rpi:~/Documents/nfc/pn532pi/examples $ nfc-poll
nfc-poll uses libnfc 1.8.0
NFC reader: PN532 board via I2C opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
FeliCa (212 kbps) target:
        ID (NFCID2): 01  16  03  00  5b  1b  d4  02
    Parameter (PAD): 01  20  22  04  27  67  4e  ff
   System Code (SC): 00  00
Waiting for card removing...nfc_initiator_target_is_present: Target Released
done.
```

And then, after that I tried ```FeliCa_card_read.py``` script and I could read my card :
```
NTAG21x R/W
-------Looking for pn532--------
writeCommand: bytearray(b'\x02')    bytearray(b'')    [0, 0, 255, 2, 254, 212, 2, 42, 0]
wait for ack at :
1622785552.5734117

0
ready at :
1622785552.5749695

_getResponseLength length frame: bytearray(b'\x00\x80\x80\x80\x80\x80')
_getResponseLength length frame: bytearray(b'\x01\x00\x00\xff\x06\xfa')
_getResponseLength length is 6
_getResponseLength writing nack: [0, 0, 255, 255, 0, 0]
readResponse read command:  3
readResponse response: bytearray(b'2\x01\x06\x07')

Found chip PN5 0x32 Firmware ver. 1.6
writeCommand: bytearray(b'2\x05\xff\x01\xff')    bytearray(b'')    [0, 0, 255, 6, 250, 212, 50, 5, 255, 1, 255, 246, 0]
wait for ack at :
1622785552.582364

0
ready at :
1622785552.5834858

_getResponseLength length frame: bytearray(b'\x00\x80\x80\x80\x80\x80')
_getResponseLength length frame: bytearray(b'\x01\x00\x00\xff\x02\xfe')
_getResponseLength length is 2
_getResponseLength writing nack: [0, 0, 255, 255, 0, 0]
readResponse read command:  33
readResponse response: bytearray(b'')

SAMConfig

writeCommand: bytearray(b'\x14\x01\x14\x01')    bytearray(b'')    [0, 0, 255, 5, 251, 212, 20, 1, 20, 1, 2, 0]
wait for ack at :
1622785552.5904863

0
ready at :
1622785552.5919495

_getResponseLength length frame: bytearray(b'\x00\x80\x80\x80\x80\x80')
_getResponseLength length frame: bytearray(b'\x01\x00\x00\xff\x02\xfe')
_getResponseLength length is 2
_getResponseLength writing nack: [0, 0, 255, 255, 0, 0]
readResponse read command:  15
readResponse response: bytearray(b'')

Waiting for an FeliCa card...
writeCommand: bytearray(b'J\x01\x01\x00\xff\xff\x01\x00')    bytearray(b'')    [0, 0, 255, 9, 247, 212, 74, 1, 1, 0, 255, 255, 1, 0, 225, 0]
wait for ack at :
1622785552.5995765

0
ready at :
1622785552.6007297

_getResponseLength length frame: bytearray(b'\x00\x80\x80\x80\x80\x80')
_getResponseLength length frame: bytearray(b'\x00\x80\x80\x80\x80\x80')
<...>
_getResponseLength length frame: bytearray(b'\x00\x80\x80\x80\x80\x80')
_getResponseLength length frame: bytearray(b'\x01\x00\x00\xff\x18\xe8')
_getResponseLength length is 24
_getResponseLength writing nack: [0, 0, 255, 255, 0, 0]
readResponse read command:  4b
readResponse response: bytearray(b'\x01\x01\x14\x01\x01\x16\x03\x00[\x1b\xd4\x02\x01 "\x04\'gN\xff\x80\x9c')

Tag number:
1

Found a card!
  IDm : b'011603005b1bd402'
  PWm: b'0120220427674eff809c'
Traceback (most recent call last):
  File "FeliCa_card_read.py", line 121, in <module>
    loop()
  File "FeliCa_card_read.py", line 85, in loop
    print("  System Code: {:x}".format(binascii.hexlify(systemCode)))
TypeError: a bytes-like object is required, not 'int'
```
Unfortunately I could make it works only once for the moment. The ```libnfc``` always works. I don't know if it can help. I'll continue testing :)

gassajor000, 2023-09-03:
Closing since this issue is old and does not contain enough information to debug. For future Remote IO Errors please open a ticket using the corresponding template and fill in all the fields.

closed 2023-09-03 by gassajor000