gassajor000/pn532pi #16: FileNotFoundError

zenetio, 2024-01-31:
I am using the SPI mode. When I run the following snippet code:
`interface = Pn532Spi(Pn532Spi.SS0_GPIO8)
pn532 = Pn532(interface)
pn532.begin()
`
the begin() method is raising the error "FileNotFoundError: [Errno 2] No such file or directory"
I checked the interface class and saw that the begin() method is "not implemented" and there is a message "it is a virtual method"
My question: how to fix this issue?
Regards.

gassajor000, 2024-04-14:
Hi Zenetio, `begin()` is implemented on the child class `Pn532Spi`. The error you mentioned is a file system error and not a `NotImplementedError`. Can you please post the full stack trace including which file it can't find?

gassajor000, 2024-08-03:
No response from requester. Closing issue as this is likely not an issue with pn532pi.

closed 2024-08-03 by gassajor000