gassajor000/pn532pi #17: [BUG] - Allow adjusting SPI bus speed to fix buggy pn532 modules

ebradbury, 2024-04-03:
### What happened?

Many people appear to be struggling to get the pn532 module to work over SPI (myself included). I'm using the elechous pn532 v3 with a rpi 5. I was only able to get it to work by decreasing the SPI bus speed in a rather hacky way.

The solution for me was adding `nfc._interface._spi.max_speed_hz = 100000` after `nfc.begin()`. Otherwise the device would never connect.

Hope that helps someone. Maybe we can make 100kz the default speed or provide an option to adjust the speed in the constructor?

### Reproducibility

Always (100% of runs)

### Steps to Reproduce

1. Connect elechous pn532 v3 module to rpi 5 SPI
2. Run SPI example
3. See that device is not detected

### Code to reproduce

```Python
# Use examples/iso14443a_uid.py
```

### Host Controller

Raspberry Pi 5

### Python Version

3.11.2

### Interface Mode

SPI

### Power Supply

Pi 3.3v or 5v

gassajor000, 2024-04-14:
Yes that does seem to be an oversight in the original arduino library. I'll can push a patch to specify a bus speed. 100 KHz seems like a pretty significant speed reduction though. Have you tried anything between 1 and 5 MHz?

gassajor000, 2024-05-05:
Haven't heard anything back so I am just going to lower the default speed to 4 MHz.

closed 2024-05-05 by gassajor000

ebradbury, 2024-05-06:
Thanks for adjusting this!

PR #19 (2024-05-05): 17 set spi bus speed
Allow setting the SPI bus speed in the Interface constructor and lower the default speed to 4 MHz.
Closes #17
commit 1bf9987f (2024-05-05): 17 set spi bus speed (#19)

* feat: Allow passing in the bus speed when creating a SPI interface.
Change default speed to 4MHz as people are reporting problems using 5MHz.
Closes #17

* build: Bump to v1.6