gassajor000/pn532pi #18: OSError: [Errno 121] Remote I/O error - (problem when calling getFirmwareVersion())

owntheweb, 2024-05-05:
### What happened?

Hello! Thanks for putting this together! Please be patient while I learn how to use a logic analyzer. I'll get started on this today. Feel free to close this ticket if I take too long. I want to get you all the info I can.

===

I'm seeing a familiar issue to other closed issues:

```
Traceback (most recent call last):
  File "/home/me/experiments/pn532.py", line 16, in <module>
    setup()
  File "/home/me/experiments/pn532.py", line 9, in setup
    versiondata = nfc.getFirmwareVersion()
  File "/home/me/.local/lib/python3.9/site-packages/pn532pi/nfc/pn532.py", line 154, in getFirmwareVersion     
    if (self._interface.writeCommand(bytearray([PN532_COMMAND_GETFIRMWAREVERSION]))):
  File "/home/me/.local/lib/python3.9/site-packages/pn532pi/interfaces/pn532i2c.py", line 57, in writeCommand  
    return self._readAckFrame()
  File "/home/me/.local/lib/python3.9/site-packages/pn532pi/interfaces/pn532i2c.py", line 158, in _readAckFrame
    responses = self._wire.transaction(reading(PN532_I2C_ADDRESS, len(PN532_ACK) + 1))
  File "/home/me/.local/lib/python3.9/site-packages/quick2wire/i2c.py", line 78, in transaction
    ioctl(self.fd, I2C_RDWR, ioctl_arg)
OSError: [Errno 121] Remote I/O error
```

I think the major difference is that I'm using a dedicated 3.3v power source: a 5V battery and/or (trying both) USB power plug stepped down with [this buck module](https://www.amazon.com/dp/B07CP4P5XJ?psc=1).

Let me see what I can do to fill in the logic analyzer data you seek. Thanks again in advance!

### Reproducibility

Always (100% of runs)

### Steps to Reproduce

1. On the pn532, set the top switch to 1 and the bottom switch to 0 to enable I2c.
2. Connect 5V to buck pin 1, pn522 to pin 2 and GND to pin 3 (see Amazon product page images for example)
3. Connect pn532 I2C and GND pins to Pi.
4. Run `sudo i2cd2tector -y 1`. See pn532 listed at 0x48.
5. Install pn532pi with `pip install pn532pi`.
6. Run "Code to reproduce" to see the error. `python helpMeRhondaHelpHelpMeRhonda.py`.

### Code to reproduce

```Python
from pn532pi import Pn532I2c, Pn532
import time

i2c = Pn532I2c(1)
nfc = Pn532(i2c)

def setup():
    nfc.begin()
    versiondata = nfc.getFirmwareVersion()

def loop():
    time.sleep(1)
    print('ping...')

if __name__ == '__main__':
    setup()
    found = loop()
    while not found:
      found = loop()
```

### Host Controller

Raspberry Pi 2 W

### Python Version

Python 3.9.2

### Interface Mode

I2C

### Power Supply

USB 5V USB battery -> AMS1117-3.3 DC 4.75-12 to 3.3V buck

### Logic Traces and other details

COMING SOON!

owntheweb, 2024-05-05:
Update: If I unplug 3.3v from the pn532, repower and start the script, I'm able to get a reading!:

Add:
```
print("Found chip PN5 {:#x} Firmware ver. {:d}.{:d}".format((versiondata >> 24) & 0xFF, (versiondata >> 16) & 0xFF,
                                                                (versiondata >> 8) & 0xFF))
```

Output 🤩:
```
Found chip PN5 0x32 Firmware ver. 1.6
```

With that, I wonder if something is happening as I'm powering up the Pi and starting Raspberry Pi OS (I removed the desktop option in raspi-config). Maybe there's a way to restart this thing on script start? I'll keep digging and will get to the logic asap (pinging friends, I bet someone has one of these already).

owntheweb, 2024-05-05:
The HSU option is working great for me with the same dedicated 3.3v supply. With that, I'll close this ticket so not to burn too much time. Regardless, if you would like assist once I get logic analyzers figured out (high on my list), please let me know!

closed 2024-05-05 by owntheweb

gassajor000, 2024-05-05:
Ok, if you do decide to debug i2c with the logic analyzer go ahead and post the traces and I'll take a look. Your comments about the pn532 and the rpi sharing a power supply echo similar issues seen by others. Originally I suspected current draw limitations but now I am wondering if there is some signal noise or power spikes causing issues.