gassajor000/pn532pi #20: Help: InDataExchange fails

iyxan23, 2024-05-15:
Hi! Thank you for this awesome project, I've been tinkering with it for a while on my RPi and it had been awesome. It works great on my MIFARE cards.

Recently I decided to develop an android app to emulate a card with HCE and let the PN532 retrieve something from it by sending an APDU SELECT command.

But I can't seem to be able to figure out the reason why PN532 seem to reject the InDataExchange command I wrote.

It returned `0x27` (`bytearray("\'")`), which according to the user manual that the problem is that "the target number is unknown".

[image]

The debugs:

[image]

My code essentially does a SAMConfiguration at the start, then does a `readPassiveTargetID` with `PN532_MIFARE_ISO14443A_106KBPS`. Then when a card is found, it will send out an `inDataExchange` with a APDU SELECT command with the AID the same as the one I set on my custom android app.

Any ideas what's going on? Am I missing something? thanks!

iyxan23, 2024-05-17:
Figured it out! Apparently I didn't set the last positional parameter of `readPassiveTargetID` to `True` which doesn't save the card id into the property `self.inListedTag` (which is later used on `inDataExchange` as a target card). That was the reason why PN532 returned a `0x27`, because the library didn't save the target id that it got from `readPassiveTargetID`.

https://github.com/gassajor000/pn532pi/blob/834b621e8b0dcc82dc3f25360a25d1f78c55848b/pn532pi/nfc/pn532.py#L362

https://github.com/gassajor000/pn532pi/blob/834b621e8b0dcc82dc3f25360a25d1f78c55848b/pn532pi/nfc/pn532.py#L415-L416

https://github.com/gassajor000/pn532pi/blob/834b621e8b0dcc82dc3f25360a25d1f78c55848b/pn532pi/nfc/pn532.py#L720-L732

I seriously cannot believe how I did not found this out for like two days.

gassajor000, 2024-05-18:
Glad you got it worked out 👍