gassajor000/pn532pi #21: Whenever I run the script it keeps through error until module disconnect

abdallaemadeldin, 2024-05-20:
### What happened?

```py from pn532pi import Pn532I2c, Pn532, pn532
import time

i2c = Pn532I2c(1)
nfc = Pn532(i2c)

def setup():
    nfc.begin()

def read_ntag213():
    while True:
        try:
            uid = nfc.readPassiveTargetID(cardbaudrate=pn532.PN532_MIFARE_ISO14443A_106KBPS)

            if uid is not None:
                print("Tag detected! UID:", [hex(i) for i in uid])

                sector = 1
                block = 0

                # Authenticate with the default key
                if nfc.mifareclassic_AuthenticateBlock(uid, sector, key_number=pn532.MIFARE_CMD_AUTH_A, key=None):
                    data = nfc.mifareclassic_ReadDataBlock(block)
                    print("Data read from NTAG213:", data)

                    nfc.mifareclassic_StopCrypto1()
                else:
                    print("Authentication failed")

            time.sleep(1)

        except Exception as e:
            print("Error:", e)
            
if __name__ == "__main__":
    setup()
    read_ntag213()
```

I'm trying to use this code to read the NTAG213 using PN532 and raspberry pi 5 and whenever I run the script it keep through `Error: 'bytearray' object cannot be interpreted as an integer` and tbh Idk how to solve this issue any recommendations, thanks.

### Reproducibility

Always (100% of runs)

### Steps to Reproduce

- mkdir pn532.py
- nano pn532.py
- copy code from prev section
- save and run `python3 pn532.py`

### Code to reproduce

```Python
from pn532pi import Pn532I2c, Pn532, pn532
import time

i2c = Pn532I2c(1)
nfc = Pn532(i2c)

def setup():
    nfc.begin()

def read_ntag213():
    while True:
        try:
            uid = nfc.readPassiveTargetID(cardbaudrate=pn532.PN532_MIFARE_ISO14443A_106KBPS)
[... 140 lines cut ...]
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer
Error: 'bytearray' object cannot be interpreted as an integer

gassajor000, 2024-08-03:
Hey sorry I didn't see this issue until now. Do you still need help with this? 
If so I am going to need more information from the cause of the exception. Add something like `print(traceback.format_exc())` to your except block and send me the output.

Carlos-ACJ, 2024-08-11:
@gassajor000 "I reproduced it using the code above on a Raspberry Pi 5 (RP5) running the official Debian 12 Lite OS. The traceback gives me:

```
Traceback (most recent call last):
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 14, in read_ntag213
    uid = nfc.readPassiveTargetID(cardbaudrate=pn532.PN532_MIFARE_ISO14443A_106KBPS)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pn532pi/nfc/pn532.py", line 420, in readPassiveTargetID
    if (self._interface.writeCommand(header)) :
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pn532pi/interfaces/pn532i2c.py", line 58, in writeCommand
    return self._readAckFrame()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pn532pi/interfaces/pn532i2c.py", line 160, in _readAckFrame
    responses = self._wire.transaction(reading(PN532_I2C_ADDRESS, len(PN532_ACK) + 1))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/quick2wire/i2c.py", line 78, in transaction
    ioctl(self.fd, I2C_RDWR, ioctl_arg)
OSError: [Errno 121] Remote I/O error

Error: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in read_ntag213
    print("Tag detected! UID:", [hex(i) for i in uid])
                                ^^^^^^^^^^^^^^^^^^^^^
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in <listcomp>
    print("Tag detected! UID:", [hex(i) for i in uid])
                                 ^^^^^^
TypeError: 'bytearray' object cannot be interpreted as an integer

Error: 'bytearray' object cannot be interpreted as an integer
Traceback (most recent call last):
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in read_ntag213
    print("Tag detected! UID:", [hex(i) for i in uid])
                                ^^^^^^^^^^^^^^^^^^^^^
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in <listcomp>
    print("Tag detected! UID:", [hex(i) for i in uid])
                                 ^^^^^^
TypeError: 'bytearray' object cannot be interpreted as an integer

Error: 'bytearray' object cannot be interpreted as an integer
Traceback (most recent call last):
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in read_ntag213
    print("Tag detected! UID:", [hex(i) for i in uid])
                                ^^^^^^^^^^^^^^^^^^^^^
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in <listcomp>
    print("Tag detected! UID:", [hex(i) for i in uid])
                                 ^^^^^^
TypeError: 'bytearray' object cannot be interpreted as an integer

Error: 'bytearray' object cannot be interpreted as an integer
```

and it keeps looping the error:

```
Traceback (most recent call last):
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in read_ntag213
    print("Tag detected! UID:", [hex(i) for i in uid])
                                ^^^^^^^^^^^^^^^^^^^^^
  File "/home/beermatic/BeermaticPHP_Station/src/python/test.py", line 17, in <listcomp>
    print("Tag detected! UID:", [hex(i) for i in uid])
                                 ^^^^^^
TypeError: 'bytearray' object cannot be interpreted as an integer

Error: 'bytearray' object cannot be interpreted as an integer
```

gassajor000, 2024-10-05:
The return type of `readPassiveTargetID()` is a tuple of (bool, bytearray) with the boolean being a success indicator.
The examples typically call this function like this to unpack the tuple into the separate components. 
`success, uid = nfc.readPassiveTargetID(...)`

closed 2024-10-05 by gassajor000