gassajor000/pn532pi #25: [IOERROR] - problem when doing i2c communication on the raspberry pi 5 using the Elechose NFC Module V3

ekinsol, 2024-12-28:
### What happened?

I have a PN532 NFC MODULE V3 connected over i2c.

sudo i2cdetect -y 1 -> Shows it connected on 0x24
nfc-poll also detects cards:

nfc-poll uses libnfc 1.8.0
NFC reader: PN532 over I2C opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  44  
       UID (NFCID1): 04  a3  0b  5a  eb  6c  80  
      SAK (SEL_RES): 00 

Couldn't get my scripts to work, so now switched to the example (ntag21x_rw.py).

When running the sample, I get this error.

```
-------Looking for PN532--------
Traceback (most recent call last):
  File "/home/p1/flowz/devicebridge/pn532pi/examples/ntag21x_rw.py", line 77, in <module>
    setup()
  File "/home/p1/flowz/devicebridge/pn532pi/examples/ntag21x_rw.py", line 39, in setup
    versiondata = nfc.getFirmwareVersion()
                  ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/p1/flowz/devicebridge/flowz/lib/python3.11/site-packages/pn532pi/nfc/pn532.py", line 183, in getFirmwareVersion
    if (self._interface.writeCommand(bytearray([PN532_COMMAND_GETFIRMWAREVERSION]))):
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/p1/flowz/devicebridge/flowz/lib/python3.11/site-packages/pn532pi/interfaces/pn532i2c.py", line 58, in writeCommand
    return self._readAckFrame()
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/p1/flowz/devicebridge/flowz/lib/python3.11/site-packages/pn532pi/interfaces/pn532i2c.py", line 160, in _readAckFrame
    responses = self._wire.transaction(reading(PN532_I2C_ADDRESS, len(PN532_ACK) + 1))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/p1/flowz/devicebridge/flowz/lib/python3.11/site-packages/quick2wire/i2c.py", line 78, in transaction
    ioctl(self.fd, I2C_RDWR, ioctl_arg)
OSError: [Errno 121] Remote I/O error
```

I can confirm that there is data (0) returned on the SMBus.

```
import smbus

bus = smbus.SMBus(1)  # I2C bus 1
address = 0x24        # Replace with your NFC device's address
try:
    data = bus.read_byte(address)
    print(f"Read byte: {data}")
except Exception as e:
    print(f"Error accessing I2C device: {e}")
```

### Reproducibility

Always (100% of runs)

### Steps to Reproduce

1. Run example ntag21x_rw.py with i2c connection set to true

### Code to reproduce

```Python
ntag21x_rw.py example
```

### Host Controller

Raspberry 5

### Python Version

3.8

### Interface Mode

I2C

### Power Supply

Wall adapter 5V

### Logic Traces and other details

Can't get sigrok to work :/

ekinsol, 2024-12-28:
Adding info:

After running ntag21x_rw.py and receiving the error, when I do nfc-poll, I get this error message:

nfc-poll uses libnfc 1.8.0
error   libnfc.chip.pn53x       Unexpected PN53x reply!
pn53x_check_communication: Input / Output Error

Running it the second time then works:

nfc-poll uses libnfc 1.8.0
NFC reader: PN532 over I2C opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  44  
       UID (NFCID1): 04  a3  0b  5a  eb  6c  80  
      SAK (SEL_RES): 00

gassajor000, 2025-02-21:
Hi ekinsol, not sure if you're still having this issue but I will need a little more information to debug. If you do have a logic analyzer, many of the $10-15 ones also work with Seleae Logic pro (free download). Most likely this is a timing issue or the controller is not seeing the ready frame sent by the pn532. I'll need a logic trace to determine which. When I get some time I can try to reproduce this with my setup.

Spuriosity1, 2025-04-16:
I'm also getting this exact error on a Rpi5. I'm fairly sure it's an issue with this library, since `libnfc` can successfully dump data off certain tags.

Hardware:
- Sparkfun pn532 module, connected over i2c
- Pi 5 (With i2c baudrate lowered to 100kHz in `/boot/firmware/config.txt`)

Diagnostics:
`i2cdetect -y 1` shows a device at address `24`

```
$ LIBNFC_LOG_LEVEL=3 nfc-list
debug	libnfc.config	key: [device.name], value: [pn532-i2c]
debug	libnfc.config	key: [device.connstring], value: [pn532_i2c:/dev/i2c-1]
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	1 device(s) defined by user
debug	libnfc.general	  #0 name: "pn532-i2c", connstring: "pn532_i2c:/dev/i2c-1"
nfc-list uses libnfc 1.8.0
debug	libnfc.general	0 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
debug	libnfc.chip.pn53x	SAMConfiguration
debug	libnfc.chip.pn53x	Timeout value: 1000
debug	libnfc.bus.i2c	TX: 00 00 ff 03 fd d4 14 01 17 00
debug	libnfc.bus.i2c	wrote 10 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.bus.i2c	TX: 00 00 ff 09 f7 d4 00 00 6c 69 62 6e 66 63 be 00
debug	libnfc.bus.i2c	wrote 16 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.bus.i2c	TX: 00 00 ff 02 fe d4 02 2a 00
debug	libnfc.bus.i2c	wrote 9 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.bus.i2c	TX: 00 00 ff 03 fd d4 12 14 06 00
debug	libnfc.bus.i2c	wrote 10 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.general	"pn532-i2c" (pn532_i2c:/dev/i2c-1) has been claimed.
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD False
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.bus.i2c	TX: 00 00 ff 0c f4 d4 06 63 02 63 03 63 0d 63 38 63 3d b0 00
debug	libnfc.bus.i2c	wrote 19 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.bus.i2c	TX: 00 00 ff 08 f8 d4 08 63 02 80 63 03 80 59 00
debug	libnfc.bus.i2c	wrote 15 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.bus.i2c	TX: 00 00 ff 04 fc d4 32 01 00 f9 00
debug	libnfc.bus.i2c	wrote 11 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.bus.i2c	TX: 00 00 ff 04 fc d4 32 01 01 f8 00
debug	libnfc.bus.i2c	wrote 11 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.bus.i2c	TX: 00 00 ff 06 fa d4 32 05 ff ff ff f8 00
debug	libnfc.bus.i2c	wrote 13 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.general	set_property_bool NP_AUTO_ISO14443_4 True
debug	libnfc.general	set_property_bool NP_FORCE_ISO14443_A True
debug	libnfc.general	set_property_bool NP_FORCE_SPEED_106 True
debug	libnfc.general	set_property_bool NP_ACCEPT_INVALID_FRAMES False
[... 271 lines cut ...]
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	InCommunicateThru
debug	libnfc.bus.i2c	TX: 00 00 ff 02 fe d4 42 ea 00
debug	libnfc.bus.i2c	wrote 9 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.bus.i2c	TX: 00 00 ff 06 fa d4 32 05 ff ff ff f8 00
debug	libnfc.bus.i2c	wrote 13 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	InRelease
debug	libnfc.bus.i2c	TX: 00 00 ff 03 fd d4 52 00 da 00
debug	libnfc.bus.i2c	wrote 10 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD False
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.bus.i2c	TX: 00 00 ff 04 fc d4 32 01 00 f9 00
debug	libnfc.bus.i2c	wrote 11 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.chip.pn53x	PowerDown
debug	libnfc.bus.i2c	TX: 00 00 ff 03 fd d4 16 f0 26 00
debug	libnfc.bus.i2c	wrote 10 bytes successfully.
debug	libnfc.chip.pn53x	PN53x ACKed
```

ekinsol, 2025-06-12:
I haven't been able to make progress on it. Nice research @antivodonik!

@gassajor000 Is this all you would need?

schmich, 2026-03-30:
I got access to the hardware again and was able to verify the fix, and I've submitted a PR. It's a mix of the two patches above. The fix doesn't reference `errno.EREMOTEIO` directly because that's only available in Python 3.11+ and I noticed this library has support for Python 3.4, so there's no sense in breaking compatibility.

My hunch was correct about the origin of the issue but Claude did the deep dive analysis which you can see in the PR comment. At a high level, the RPi 5 uses a different I2C chip and kernel driver from the RPi 4. The RPi 5 driver returns `EREMOTEIO` instead of `EIO` when it gets a `NAK` from the PN532 when polling which previously went unhandled. The fix is to treat `EREMOTEIO` like `EIO`.