gassajor000/pn532pi #26: Add support for MicroPython

fabiomolinar, 2025-03-26:
Hello there,

First of all, I just wanted to say that I am really grateful for the work that has been done here! Congrats to the team. I have spent hours trying to find a decent PN532 Python package and most of the stuff I found covered only reading tags' UIDs. This is the first time I found something that goes way beyond. Thank you very much for that!

As for this ticket's objective, I would like to ask if there is any chance or plan to support projects using [MicroPython](https://micropython.org/)? I have gone through the code and if I understand it correctly, your code depends on `quick2wire` for the interface communication. On `MicroPython`, that is handled by the language itself using dedicated modules (e.g., for *I2C*, one would `from machine import I2C`).

Best regards,
Fabio

fabiomolinar, 2025-03-26:
Just asked copilot to give it a go and try to come up with an interface that would support MicroPython instead. This is what it gave me:

```
import time
from pn532pi.nfc.pn532_log import DMSG
from machine import I2C, Pin
import errno

from pn532pi.interfaces.pn532Interface import Pn532Interface, PN532_PREAMBLE, PN532_STARTCODE1, PN532_STARTCODE2, PN532_HOSTTOPN532, \
    PN532_INVALID_FRAME, PN532_POSTAMBLE, PN532_PN532TOHOST, PN532_ACK_WAIT_TIME, PN532_TIMEOUT, \
    PN532_INVALID_ACK

PN532_I2C_ADDRESS = (0x48 >> 1)

class Pn532I2c(Pn532Interface):
    RPI_BUS0 = 0
    RPI_BUS1 = 1

    def __init__(self, bus: int, scl_pin: int, sda_pin: int):
        assert bus in [self.RPI_BUS0, self.RPI_BUS1], "Bus number must be 1 or 0"
        self._wire = None
        self._bus = bus
        self._scl_pin = scl_pin
        self._sda_pin = sda_pin
        self._command = 0

    def begin(self):
        self._wire = I2C(self._bus, scl=Pin(self._scl_pin), sda=Pin(self._sda_pin))
        time.sleep(1)

    def wakeup(self):
        time.sleep(.05) # wait for all ready to manipulate pn532
        self._wire.writeto(PN532_I2C_ADDRESS, bytes([0]))

    def writeCommand(self, header: bytearray, body: bytearray = bytearray()):
        self._command = header[0]
        data_out = [PN532_PREAMBLE, PN532_STARTCODE1, PN532_STARTCODE2]

        length = len(header) + len(body) + 1 # length of data field: TFI + DATA
        data_out.append(length)
        data_out.append((~length & 0xFF) + 1) # checksum of length

        data_out.append(PN532_HOSTTOPN532)
        dsum = PN532_HOSTTOPN532 + sum(header) + sum(body)  # sum of TFI + DATA

        data_out += list(header)
        data_out += list(body)
        checksum = ((~dsum & 0xFF) + 1) & 0xFF # checksum of TFI + DATA

        data_out += [checksum, PN532_POSTAMBLE]

        DMSG("writeCommand: {}    {}    {}".format(header, body, data_out))

        try:
            # send data
            self._wire.writeto(PN532_I2C_ADDRESS, bytes(data_out))
        except Exception as e:
            DMSG(e)
            DMSG("\nToo many data to send, I2C doesn't support such a big packet\n")  # I2C max packet: 32 bytes
            return PN532_INVALID_FRAME

        return self._readAckFrame()

    def _getResponseLength(self, timeout: int):
        PN532_NACK = [0, 0, 0xFF, 0xFF, 0, 0]
        timer = 0

        while 1:
            self._wire.writeto(PN532_I2C_ADDRESS, bytes([]))
            data = self._wire.readfrom(PN532_I2C_ADDRESS, 6)
[... 97 lines cut ...]
            except OSError as e:
                if e.errno != errno.EIO:
                    raise   # Reraise the error   
                # Otherwise do nothing, sleep and try again
            
            time.sleep(.001)    # sleep 1 ms
            t+=1
        else:
            DMSG("Time out when waiting for ACK\n")
            return PN532_TIMEOUT

        DMSG("ready at : ")
        DMSG(time.time())
        DMSG('\n')

        ackBuf = list(data[1:])

        if ackBuf != PN532_ACK:
            DMSG("Invalid ACK {}\n".format(ackBuf))
            return PN532_INVALID_ACK

        return 0
```

I didn't really analyze it. But once my PN532 board arrives, I'll give it a go.

fabiomolinar, 2025-03-26:
By the way, I forgot to mention, I am using MicroPython because I am developing on an ESP32 board. Do you think the only affected modules would be `pn532pi.interfaces`?

gassajor000, 2025-03-27:
Hey, thanks for suggesting this. There is no roadmap for micro Python
support at the moment but you're welcome to take a stab at it and open a PR
if you get it working. The code is setup so that you should just have to
implement the i2c interface. I'll have to check if there are any other
dependencies that would need to be addressed.
Thanks,
-Jordan

fabiomolinar, 2025-03-27:
I'll give it a try in the next couple of days. Let me know if you think any other dependencies would be impacted. Hopefully not. :)
Regards, 
Fabio

gassajor000, 2025-03-28:
After a quick look through the only thing that I think might cause problems is the typing. I've never got that to work will on microcontrollers

fabiomolinar, 2025-03-28:
Have you tried [this option](https://micropython-stubs.readthedocs.io/en/main/typing_mpy.html)? This is from the same developer that created the stubs I am using. I'll give it a shot as well to see if it gets the typing module fixed as well.

fabiomolinar, 2025-03-28:
Regarding the below:

> Have you tried [this option](https://micropython-stubs.readthedocs.io/en/main/typing_mpy.html)? This is from the same developer that created the stubs I am using. I'll give it a shot as well to see if it gets the typing module fixed as well.

I just gave it a try. I didn't try it extensively, but I was able to import it into one of my modules and add typing to some of my functions and it compiled and ran without a problem in my ESP32 board.

fabiomolinar, 2025-03-28:
It's a little late for me but today I was able to read the firmware on my PN532 using my code. Will test the rest maybe tomorrow.

I did have to do a change to the `nfc.pn532` module in `pn532.py`. Line 199, the original is `return int.from_bytes(response, byteorder='big')`. However, when running as such I get a `TypeError`. The `int.from_bytes` implementation on **MicroPython** does not accept a `byteorder` keyword argument. I had to modify it to `return int.from_bytes(response, 'big')` to get it to run and read the firmware. 

Will keep testing soon.

fabiomolinar, 2025-03-29:
Went through [this example](https://github.com/gassajor000/pn532pi/blob/master/examples/readMifare.py) today and it worked flawlessly! The output I got on the terminal using a test tag:

```
_getResponseLength length frame: b'\x00\x80\x80\x80\x80\x80'
_getResponseLength length frame: b'\x00\x80\x80\x80\x80\x80'
_getResponseLength length frame: b'\x00\x80\x80\x80\x80\x80'
_getResponseLength length frame: b'\x01\x00\x00\xff\x0c\xf4'
_getResponseLength length is 12
_getResponseLength writing nack: [0, 0, 255, 255, 0, 0]
readResponse read command:  4b
readResponse response: b'\x01\x01\x00\x04\x08\x04V\x83\xc8\x01'

ATQA: 0x
4
SAK: 0x
8

Found an ISO14443A card
UID Length: 4
UID Value: b'5683c801'
Seems to be a Mifare Classic card (4 byte UID)
Trying to authenticate block 4 with default KEYA value
writeCommand: bytearray(b'@\x01`\x04\xff\xff\xff\xff\xff\xffV\x83\xc8\x01') bytearray(b'') [0, 0, 255, 15, 241, 212, 64, 1, 96, 4, 255, 255, 255, 255, 255, 255, 86, 131, 200, 1, 235, 0]
wait for ack at :
488
ready at :
488
_getResponseLength length frame: b'\x00\x80\x80\x80\x80\x80'
_getResponseLength length frame: b'\x01\x00\x00\xff\x03\xfd'
_getResponseLength length is 3
_getResponseLength writing nack: [0, 0, 255, 255, 0, 0]
readResponse read command:  41
readResponse response: b'\x00'

Sector 1 (Blocks 4..7) has been authenticated
Trying to read 16 bytes from block
4
writeCommand: bytearray(b'@\x010\x04') bytearray(b'') [0, 0, 255, 5, 251, 212, 64, 1, 48, 4, 183, 0]
wait for ack at :
488
ready at :
488
_getResponseLength length frame: b'\x00\x80\x80\x80\x80\x80'
_getResponseLength length frame: b'\x00\x80\x80\x80\x80\x80'
_getResponseLength length frame: b'\x00\x80\x80\x80\x80\x80'
_getResponseLength length frame: b'\x01\x00\x00\xff\x13\xed'
_getResponseLength length is 19
_getResponseLength writing nack: [0, 0, 255, 255, 0, 0]
readResponse read command:  41
readResponse response: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Reading Block 4: b'00000000000000000000000000000000'
True
```

fabiomolinar, 2025-03-29:
I'll paste the code for the I2C interface for **MicroPython** below for a quick discussion before sending a PR. 

My PR could look like this:

```
pn532pi/
│── interfaces/
│   ├── micropython/
│   │   ├── __init__.py
│   │   ├── pn532i2c.py
```

Where the contents of the `pn532i2c.py` are shown below. But as you can see from the below, my code is almost a copy of your code for Raspberry Pi. Hence, I wonder if a better approach would be to instead create a `Pn532I2CInterface` to avoid repeating the code again, leaving the interface only for the interface handling. But I am not sure about the other interfaces, so I can't tell what is the best approach here. 

Let me know what you think.

```python
import errno
import time
import machine
from pn532pi.nfc.pn532_log import DMSG
from pn532pi.interfaces.pn532Interface import (
    Pn532Interface, PN532_PREAMBLE, PN532_STARTCODE1, PN532_STARTCODE2,
    PN532_HOSTTOPN532, PN532_INVALID_FRAME, PN532_POSTAMBLE, PN532_PN532TOHOST,
    PN532_ACK_WAIT_TIME, PN532_TIMEOUT, PN532_INVALID_ACK
)

PN532_I2C_ADDRESS = (0x48 >> 1)

class Pn532I2c(Pn532Interface):
    def __init__(self, scl_pin: int, sda_pin: int):
        """
        Initialize the PN532 I2C interface using MicroPython's machine.I2C.

        Args:
            scl_pin (int): GPIO pin for I2C SCL.
            sda_pin (int): GPIO pin for I2C SDA.
            freq (int): I2C frequency (default: 100kHz).
        """
        self._i2c = machine.I2C(0, scl=machine.Pin(scl_pin), sda=machine.Pin(sda_pin))
        self._command = 0

    def begin(self):
        """
        Initialize the I2C interface. This is a placeholder for compatibility.
        """
        time.sleep(1)  # Allow time for the PN532 to initialize.

    def wakeup(self):
        """
        Wake up the PN532 by sending a dummy write.
        """
        time.sleep(0.05)  # Wait for the PN532 to be ready.        
        self._i2c.writeto(PN532_I2C_ADDRESS, b'\x00')

    def writeCommand(self, header: bytearray, body: bytearray = bytearray()):
        """
        Write a command to the PN532.

        Args:
            header (bytearray): Command header.
            body (bytearray): Command body (optional).

        Returns:
            int: Status of the operation.
        """
        self._command = header[0]
        data_out = [PN532_PREAMBLE, PN532_STARTCODE1, PN532_STARTCODE2]

        length = len(header) + len(body) + 1  # Length of data field: TFI + DATA
[... 146 lines cut ...]
                data = self._i2c.readfrom(PN532_I2C_ADDRESS, len(PN532_ACK) + 1)
                if (data[0] & 1):   # Check first byte --- status                    
                    break # PN532 is ready
            except OSError as e:
                if e.errno != errno.EIO:
                    raise   # Reraise the error  
                # Otherwise do nothing, sleep and try again
                
            time.sleep(.001)    # sleep 1 ms
            t+=1
        else:
            DMSG("Time out when waiting for ACK\n")
            return PN532_TIMEOUT
        
        DMSG("ready at : ")
        DMSG(time.time())

        ackBuf = list(data[1:])

        if ackBuf != PN532_ACK:
            DMSG("Invalid ACK {}\n".format(ackBuf))
            return PN532_INVALID_ACK

        return 0
```

gassajor000, 2025-03-31:
Awesome 👍
Yeah I think the preferred option would be to create an interface file like
the others. If you open a PR I can test it out when I get some free time
Thanks,
-Jordan

fabiomolinar, 2025-03-31:
Created PR #27. Let me know what you think or if you need any modifications.