gassajor000/pn532pi #28: Add support for asynchronous calls

fabiomolinar, 2025-04-04:
Feature request

In its current form, the package supports only synchronous calls. It would be great if the package could support asynchronous calls as well.

gassajor000, 2025-04-25:
Hmm that might get a bit complicated. Especially considering we are in the process of adding micropython support. How would you suggest implementing that?

fabiomolinar, 2025-04-26:
I changed it in my code by mostly changing the functions I am using to be `async` and to `await` any other function it may use (these also need to be made `async`). 

See some examples below.

```python
    async def begin(self):
        """
        Setups the HW
        """
        await self._interface.begin()
        await self._interface.wakeup()

    async def getFirmwareVersion(self) -> int:
        """
        Checks the firmware version of the PN5xx chip

        See https://www.nxp.com/docs/en/user-guide/141520.pdf page 73

        :returns:  The chip's firmware version and ID
        """
        if (await self._interface.writeCommand(bytearray([PN532_COMMAND_GETFIRMWAREVERSION]))):
            return 0

        # read data packet
        status, response = await self._interface.readResponse()
        if (status < 0):
            return 0

        # response = self.pn532_packetbuffer[0]
        # response <<= 8
        # response |= self.pn532_packetbuffer[1]
        # response <<= 8
        # response |= self.pn532_packetbuffer[2]
        # response <<= 8
        # response |= self.pn532_packetbuffer[3]

        return int.from_bytes(response, 'big')
```

fabiomolinar, 2025-04-26:
By making them `async`, I can optimize my controller to run some other tasks in parallel while the board is reading data.